simulton/QSchematic #16: error refers to serialize.hpp when gpds/serialize.h exists instead

ckqee, 2020-02-18:
Hi,
Seeing the following issue:

QSchematic/qschematic/items/wirenet.h:6:10: fatal error: gpds/serialize.hpp: No such file or directory
 #include <gpds/serialize.hpp>

The recommended name is which one please?

Tectu, 2020-02-18:
Hi,

As mentioned in #17 I think the issue is that the QSchematic repository currently refers to an old/outdated commit of the GPDS submodule repo.
Try to check out the latest develop branch of the GPDS submodule and try to perform a clean build.

I'll bump the GPDS submodule commit once we confirmed that this is the problem.

Tectu, 2020-02-24:
As mentioned before this boiled down to an issue where the GPDS master branch was too far behind. We've updated the GPDS master branch and we were able to build QSchematic on a clean machine. Keep in mind that QSchematic depends on the GPDS library for (de)serialization. You have to build & install the GPDS library.

We've added very basic build instructions to the readme. As everything is cmake this should be pretty straight forward but I don't doubt that a bit more might be helpful.

Please don't hesitate to ask if you run into other issues.

Tectu, 2020-03-04:
Any news on this?

ckqee, 2020-03-07:
Seeing the following - what is the correct way/recommendation to install gpds ?

```
user1@machine:~/github/simulton/QSchematic$ git pull
remote: Enumerating objects: 82, done.
remote: Counting objects: 100% (82/82), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 82 (delta 53), reused 62 (delta 40), pack-reused 0
Unpacking objects: 100% (82/82), 43.07 KiB | 436.00 KiB/s, done.
From https://github.com/simulton/QSchematic
   ee829be..194552b  master                 -> origin/master
   ee829be..1bc3b56  develop                -> origin/develop
   3ee3a85..ffc4d25  feature/move_wire_code -> origin/feature/move_wire_code
Updating ee829be..194552b
Fast-forward
 readme.md | 19 +++++++++++++++++++
 1 file changed, 19 insertions(+)
user1@machine:~/github/simulton/QSchematic$ cd build/
user1@machine:~/github/simulton/QSchematic/build$ make
[  2%] Automatic MOC and UIC for target qschematic-objs
[  2%] Built target qschematic-objs_autogen
[  5%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/qschematic-objs_autogen/mocs_compilation.cpp.o
In file included from /home/user1/github/simulton/QSchematic/qschematic/items/connector.h:3,
                 from /home/user1/github/simulton/QSchematic/build/qschematic/qschematic-objs_autogen/HU7TA6T7T7/moc_connector.cpp:9,
                 from /home/user1/github/simulton/QSchematic/build/qschematic/qschematic-objs_autogen/mocs_compilation.cpp:2:
/home/user1/github/simulton/QSchematic/qschematic/items/item.h:5:10: fatal error: gpds/serialize.hpp: No such file or directory
    5 | #include <gpds/serialize.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
qschematic/CMakeFiles/qschematic-objs.dir/build.make:62: recipe for target 'qschematic/CMakeFiles/qschematic-objs.dir/qschematic-objs_autogen/mocs_compilation.cpp.o' failed
make[2]: *** [qschematic/CMakeFiles/qschematic-objs.dir/qschematic-objs_autogen/mocs_compilation.cpp.o] Error 1
CMakeFiles/Makefile2:176: recipe for target 'qschematic/CMakeFiles/qschematic-objs.dir/all' failed
make[1]: *** [qschematic/CMakeFiles/qschematic-objs.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

ckqee, 2020-03-07:
I'll paste the gpds build stuff here -- this should give you what kind of problems I am having

```
user1@machine:~/github/simulton/gpds/build$ cmake gpds-static ..
-- The CXX compiler identification is GNU 9.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/gpds/build/test/doctest
Scanning dependencies of target doctest
[ 12%] Creating directories for 'doctest'
[ 25%] Performing download step (git clone) for 'doctest'
-- doctest download command succeeded.  See also /home/user1/github/simulton/gpds/build/test/doctest/doctest-prefix/src/doctest-stamp/doctest-download-*.log
[ 37%] No patch step for 'doctest'
[ 50%] No update step for 'doctest'
[ 62%] No configure step for 'doctest'
[ 75%] No build step for 'doctest'
[ 87%] No install step for 'doctest'
[100%] Completed 'doctest'
[100%] Built target doctest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/gpds/build
user1@machine:~/github/simulton/gpds/build$ make -j
Scanning dependencies of target gpds-objs
[  6%] Building CXX object lib/CMakeFiles/gpds-objs.dir/src/archiver_xml.cpp.o
[  6%] Building CXX object lib/CMakeFiles/gpds-objs.dir/src/container.cpp.o
[  9%] Building CXX object lib/CMakeFiles/gpds-objs.dir/src/value.cpp.o
[ 12%] Building CXX object lib/CMakeFiles/gpds-objs.dir/3rdparty/tinyxml2/tinyxml2.cpp.o
[ 16%] Building CXX object lib/CMakeFiles/gpds-objs.dir/src/attributes.cpp.o
In file included from /home/user1/github/simulton/gpds/lib/include/container.hpp:5,
                 from /home/user1/github/simulton/gpds/lib/src/container.cpp:1:
/home/user1/github/simulton/gpds/lib/include/value.hpp:38:20: error: declaration of ?gpds::attributes gpds::value::attributes? changes meaning of ?attributes? [-fpermissive]
   38 |         attributes attributes;
      |                    ^~~~~~~~~~
In file included from /home/user1/github/simulton/gpds/lib/include/value.hpp:10,
                 from /home/user1/github/simulton/gpds/lib/include/container.hpp:5,
                 from /home/user1/github/simulton/gpds/lib/src/container.cpp:1:
/home/user1/github/simulton/gpds/lib/include/attributes.hpp:13:23: note: ?attributes? declared here as ?class gpds::attributes?
   13 |     class GPDS_EXPORT attributes
      |                       ^~~~~~~~~~
In file included from /home/user1/github/simulton/gpds/lib/src/value.cpp:1:
/home/user1/github/simulton/gpds/lib/include/value.hpp:38:20: error: declaration of ?gpds::attributes gpds::value::attributes? changes meaning of ?attributes? [-fpermissive]
   38 |         attributes attributes;
      |                    ^~~~~~~~~~
.
.
.
.
/home/user1/github/simulton/gpds/lib/include/attributes.hpp:13:23: note: ?attributes? declared here as ?class gpds::attributes?
   13 |     class GPDS_EXPORT attributes
      |                       ^~~~~~~~~~
In file included from /home/user1/github/simulton/gpds/lib/include/serialize.hpp:4,
                 from /home/user1/github/simulton/gpds/lib/include/archiver.hpp:6,
                 from /home/user1/github/simulton/gpds/lib/include/archiver_xml.hpp:5,
                 from /home/user1/github/simulton/gpds/lib/src/archiver_xml.cpp:2:
/home/user1/github/simulton/gpds/lib/include/container.hpp:16:20: error: declaration of ?gpds::attributes gpds::container::attributes? changes meaning of ?attributes? [-fpermissive]
   16 |         attributes attributes;
      |                    ^~~~~~~~~~
In file included from /home/user1/github/simulton/gpds/lib/include/value.hpp:10,
                 from /home/user1/github/simulton/gpds/lib/include/container.hpp:5,
                 from /home/user1/github/simulton/gpds/lib/include/serialize.hpp:4,
                 from /home/user1/github/simulton/gpds/lib/include/archiver.hpp:6,
                 from /home/user1/github/simulton/gpds/lib/include/archiver_xml.hpp:5,
                 from /home/user1/github/simulton/gpds/lib/src/archiver_xml.cpp:2:
/home/user1/github/simulton/gpds/lib/include/attributes.hpp:13:23: note: ?attributes? declared here as ?class gpds::attributes?
   13 |     class GPDS_EXPORT attributes
      |                       ^~~~~~~~~~
lib/CMakeFiles/gpds-objs.dir/build.make:62: recipe for target 'lib/CMakeFiles/gpds-objs.dir/src/archiver_xml.cpp.o' failed
make[2]: *** [lib/CMakeFiles/gpds-objs.dir/src/archiver_xml.cpp.o] Error 1
CMakeFiles/Makefile2:213: recipe for target 'lib/CMakeFiles/gpds-objs.dir/all' failed
make[1]: *** [lib/CMakeFiles/gpds-objs.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
user1@machine:~/github/simulton/gpds/build$ 

```

ckqee, 2020-03-08:
My versions of cmake, g++ and clang
```
user1@machine:~/github/simulton/gpds/build$ which g++
/usr/bin/g++
user1@machine:~/github/simulton/gpds/build$ cmake -version
cmake version 3.16.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
user1@machine:~/github/simulton/gpds/build$ g++ --version
g++ (Ubuntu 9.2.1-17ubuntu1~18.04.1) 9.2.1 20191102
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
user1@machine:~/github/simulton/gpds/build$ which clang
/usr/bin/clang
user1@machine:~/github/simulton/gpds/build$ clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

```

ckqee, 2020-03-11:
Did the above help?
Do you know what could help me debug this further?

Tectu, 2020-03-12:
I'm really not sure what the issue is here.
Our main development platform uses cmake 3.15 and GCC 9.2 - basically what you use. Our continuous integration system also compiles with clang without any issues.

We use this library successfully on Windows, Linux and BSD in a multitude of different projects & products.

My plan was to fire up a VM and compiling it manually on there but I don't expect any results different from what our developers see daily or what the continuous integration system does every day.

Tectu, 2020-03-12:
GPDS commig 5a919735efbc6f47bca034e72f2352c8ee28c975 or later should now build without any issues for you.
Please confirm.

ckqee, 2020-03-14:
Thank you! 
I can build GPDS version cleanly and install it - I did the following steps:

```
git clone https://github.com/simulton/gpds
cd gpds
mkdir build
cd build
cmake ..
make gpds-static
sudo make install
```
Seeing issues with Qschematic demo being build - I'll log a new issue for that.

closed 2020-03-14 by Tectu