simulton/QSchematic #17: connector.cpp:79:73: error: request for member ‘value’ in ‘(& container)->gpds::container::get_value<gpds::container*>(std::__cxx11::basic_string<char>(((const char*)"item"), std::allocator<char>()), 0)’, which is of pointer type ‘gpds::container*’ (maybe you meant to use ‘->’ ?)

ckqee, 2020-02-18:
Trying to compile this - coming across many errors - is this build-able?
Can you provide instructions to help compile and build this please?

```
nix_directory_path/github/simulton/QSchematic/qschematic/items/connector.cpp: In member function ‘virtual void QSchematic::Connector::from_container(const gpds::container&)’:
nix_directory_path/github/simulton/QSchematic/qschematic/items/connector.cpp:79:73: error: request for member ‘value’ in ‘(& container)->gpds::container::get_value<gpds::container*>(std::__cxx11::basic_string<char>(((const char*)"item"), std::allocator<char>()), 0)’, which is of pointer type ‘gpds::container*’ (maybe you meant to use ‘->’ ?)
     Item::from_container(*container.get_value<gpds::container*>("item").value());
                                                                         ^~~~~
nix_directory_path/github/simulton/QSchematic/qschematic/items/connector.cpp:80:83: error: request for member ‘value_or’ in ‘(& container)->gpds::container::get_value<int>(std::__cxx11::basic_string<char>(((const char*)"snap_policy"), std::allocator<char>()), 0)’, which is of non-class type ‘int’
     setSnapPolicy(static_cast<SnapPolicy>(container.get_value<int>("snap_policy").value_or(Anywhere)));
                                                                                   ^~~~~~~~
nix_directory_path/github/simulton/QSchematic/qschematic/items/connector.cpp:81:77: error: request for member ‘value_or’ in ‘(& container)->gpds::container::get_value<bool>(std::__cxx11::basic_string<char>(((const char*)"force_text_direction"), std::allocator<char>()), false)’, which is of non-class type ‘bool’
     setForceTextDirection(container.get_value<bool>("force_text_direction").value_or(false));
                                                                             ^~~~~~~~
nix_directory_path/github/simulton/QSchematic/qschematic/items/connector.cpp:82:88: error: request for member ‘value_or’ in ‘(& container)->gpds::container::get_value<int>(std::__cxx11::basic_string<char>(((const char*)"text_direction"), std::allocator<char>()), 0)’, which is of non-class type ‘int’
     _textDirection = static_cast<Direction>(container.get_value<int>("text_direction").value_or(LeftToRight));
                                                                                        ^~~~~~~~
nix_directory_path/github/simulton/QSchematic/qschematic/items/connector.cpp:83:76: error: request for member ‘value’ in ‘(& container)->gpds::container::get_value<gpds::container*>(std::__cxx11::basic_string<char>(((const char*)"label"), std::allocator<char>()), 0)’, which is of pointer type ‘gpds::container*’ (maybe you meant to use ‘->’ ?)
     _label->from_container(*container.get_value<gpds::container*>("label").value());
                                                                            ^~~~~
qschematic/CMakeFiles/qschematic-objs.dir/build.make:205: recipe for target 'qschematic/CMakeFiles/qschematic-objs.dir/items/connector.cpp.o' failed
make[2]: *** [qschematic/CMakeFiles/qschematic-objs.dir/items/connector.cpp.o] Error 1
CMakeFiles/Makefile2:176: recipe for target 'qschematic/CMakeFiles/qschematic-objs.dir/all' failed
make[1]: *** [qschematic/CMakeFiles/qschematic-objs.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

Tectu, 2020-02-18:
Hi,

This is definitely buildable. We use the library in several client applications and it's also part of our continuous integration system (build-badge can be found in the readme).

Did you check out the GPDS submodule repository?
Writing this it might be possible that the QSchematic repository refers to an old commit of the GPDS submodule dependency. Can you checkout the latest `develop` branch of the GPDS repository and try to build again?

In general this should build just like any other cmake project. We should definitely add some more documentation now that it gets used a bit more tho.

ckqee, 2020-02-19:
Hi,
I cannot seem to replicate my steps. I fall over way too earlier now.
I'll see what I can do later on
Thank you!

Tectu, 2020-02-24:
Duplicate of #16.

closed 2020-02-24 by Tectu