simulton/QSchematic #18: netlistviewer.h:20:40: error: expected ‘;’ at end of member declaration    20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )

ckqee, 2020-03-14:
Initial problem I was seeing is that I cannot get the compiler to find the commandbase.h file

```
user1@machine:~/github/simulton/QSchematic/demo/build$ make qschematic-demo
Scanning dependencies of target qschematic-demo
[  5%] Building CXX object CMakeFiles/qschematic-demo.dir/commands/commandnodeaddconnector.o
In file included from /home/user1/github/simulton/QSchematic/demo/commands/commandnodeaddconnector.cpp:4:
/home/user1/github/simulton/QSchematic/demo/commands/commandnodeaddconnector.h:3:10: fatal error: commands/commandbase.h: No such file or directory
    3 | #include <commands/commandbase.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/qschematic-demo.dir/build.make:88: recipe for target 'CMakeFiles/qschematic-demo.dir/commands/commandnodeaddconnector.o' failed
make[3]: *** [CMakeFiles/qschematic-demo.dir/commands/commandnodeaddconnector.o] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/qschematic-demo.dir/all' failed
make[2]: *** [CMakeFiles/qschematic-demo.dir/all] Error 2
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/qschematic-demo.dir/rule' failed
make[1]: *** [CMakeFiles/qschematic-demo.dir/rule] Error 2
Makefile:118: recipe for target 'qschematic-demo' failed
make: *** [qschematic-demo] Error 2
```
I edited the file 

```
QSchematic/demo/commands/commandnodeaddconnector.h
```
Such that I pointed to the include directory:

```
#include "/usr/local/include/qschematic/commands/commandbase.h"
```
It then goes pass that error then I get another error expected ';' see below:

```
user1@machine:~/github/simulton/QSchematic/demo/build$  make
Scanning dependencies of target qschematic-demo
[  5%] Building CXX object CMakeFiles/qschematic-demo.dir/commands/commandnodeaddconnector.o
[ 10%] Building CXX object CMakeFiles/qschematic-demo.dir/items/customitemfactory.o
[ 15%] Building CXX object CMakeFiles/qschematic-demo.dir/items/fancywire.o
[ 21%] Building CXX object CMakeFiles/qschematic-demo.dir/items/flowend.o
[ 26%] Building CXX object CMakeFiles/qschematic-demo.dir/items/flowstart.o
[ 31%] Building CXX object CMakeFiles/qschematic-demo.dir/items/operation.o
[ 36%] Building CXX object CMakeFiles/qschematic-demo.dir/items/operationconnector.o
[ 42%] Building CXX object CMakeFiles/qschematic-demo.dir/items/operationdemo1.o
[ 47%] Building CXX object CMakeFiles/qschematic-demo.dir/itemslibrary/itemslibrarymodel.o
[ 52%] Building CXX object CMakeFiles/qschematic-demo.dir/itemslibrary/itemslibraryview.o
[ 57%] Building CXX object CMakeFiles/qschematic-demo.dir/itemslibrary/itemsslibrarywidget.o
[ 63%] Building CXX object CMakeFiles/qschematic-demo.dir/netlistviewer/netlistviewer.o
In file included from /home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewer.cpp:2:
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewer.h:20:40: error: ISO C++ forbids declaration of ‘Q_DISABLE_COPY_MOVE’ with no type [-fpermissive]
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewer.h:20:40: error: expected ‘;’ at end of member declaration
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
      |                                         ;
In file included from /home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewer.cpp:4:
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewerview.h:13:44: error: ISO C++ forbids declaration of ‘Q_DISABLE_COPY_MOVE’ with no type [-fpermissive]
   13 |     Q_DISABLE_COPY_MOVE( NetlistViewerView )
      |                                            ^
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewerview.h:13:44: error: expected ‘;’ at end of member declaration
   13 |     Q_DISABLE_COPY_MOVE( NetlistViewerView )
      |                                            ^
      |                                             ;
CMakeFiles/qschematic-demo.dir/build.make:231: recipe for target 'CMakeFiles/qschematic-demo.dir/netlistviewer/netlistviewer.o' failed
make[2]: *** [CMakeFiles/qschematic-demo.dir/netlistviewer/netlistviewer.o] Error 1
CMakeFiles/Makefile2:75: recipe for target 'CMakeFiles/qschematic-demo.dir/all' failed
make[1]: *** [CMakeFiles/qschematic-demo.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Tectu, 2020-03-14:
Did you successfully build and install the `qschematic-static` and/or `qschematic-shared` library targets before trying to build the demo?

The demo application pulls in the `QSchematic` cmake package just to demonstrate how any client application would do this. Therefore, it needs to be installed first.

ckqee, 2020-03-17:
I retraced my steps and did a clean build and install - hence removing the edit I had done.
It feels as if I am missing a step - not sure why I am getting this error
```
netlistviewer.h:20:40: error: ISO C++ forbids declaration of ‘Q_DISABLE_COPY_MOVE’ with no type [-fpermissive]
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
```

The error message in context:

```
user1@machine:~/github/simulton/QSchematic$ mkdir build
user1@machine:~/github/simulton/QSchematic$ cd !$
cd build
user1@machine:~/github/simulton/QSchematic/build$ cmake ..
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/build
user1@machine:~/github/simulton/QSchematic/build$ make
Scanning dependencies of target qschematic-objs_autogen
[  2%] Automatic MOC and UIC for target qschematic-objs
[  2%] Built target qschematic-objs_autogen
Scanning dependencies of target qschematic-objs
[  5%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/qschematic-objs_autogen/mocs_compilation.cpp.o
[  8%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandbase.cpp.o
[ 11%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemadd.cpp.o
[ 14%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemmove.cpp.o
[ 17%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemremove.cpp.o
[ 20%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemvisibility.cpp.o
[ 23%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandlabelrename.cpp.o
[ 26%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandnoderesize.cpp.o
[ 29%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandnoderotate.cpp.o
[ 32%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandwirenetrename.cpp.o
[ 35%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandwirepointmove.cpp.o
[ 38%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/connector.cpp.o
[ 41%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/item.cpp.o
[ 44%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/itemfactory.cpp.o
[ 47%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/itemmimedata.cpp.o
[ 50%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/label.cpp.o
[ 52%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/line.cpp.o
[ 55%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/node.cpp.o
[ 58%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/splinewire.cpp.o
[ 61%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/wire.cpp.o
[ 64%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/wirenet.cpp.o
[ 67%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/wirepoint.cpp.o
[ 70%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/items/wireroundedcorners.cpp.o
[ 73%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/scene.cpp.o
[ 76%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/settings.cpp.o
[ 79%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/utils.cpp.o
[ 82%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/view.cpp.o
[... 123 lines cut ...]
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewer.h:20:40: error: expected ‘;’ at end of member declaration
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
      |                                         ;
In file included from /home/user1/github/simulton/QSchematic/build/demo/qschematic-demo_autogen/DRL6IFWHKL/moc_netlistviewerview.cpp:9,
                 from /home/user1/github/simulton/QSchematic/build/demo/qschematic-demo_autogen/mocs_compilation.cpp:15:
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewerview.h:13:44: error: ISO C++ forbids declaration of ‘Q_DISABLE_COPY_MOVE’ with no type [-fpermissive]
   13 |     Q_DISABLE_COPY_MOVE( NetlistViewerView )
      |                                            ^
[... cut ...]
make[3]: *** [demo/CMakeFiles/qschematic-demo.dir/qschematic-demo_autogen/mocs_compilation.cpp.o] Error 1
CMakeFiles/Makefile2:286: recipe for target 'demo/CMakeFiles/qschematic-demo.dir/all' failed
make[2]: *** [demo/CMakeFiles/qschematic-demo.dir/all] Error 2
CMakeFiles/Makefile2:293: recipe for target 'demo/CMakeFiles/qschematic-demo.dir/rule' failed
make[1]: *** [demo/CMakeFiles/qschematic-demo.dir/rule] Error 2
Makefile:242: recipe for target 'qschematic-demo' failed
make: *** [qschematic-demo] Error 2

```

Tectu, 2020-03-17:
Unfortunately, I am unable to reproduce this.

I am able to build both the library & the demo successfully on:
- Linux 5.4.23 64-Bit using GCC 9.2.1
- Linux 5.4.23 64-Bit using clang 9.0.1
- FreeBSD 12.1 64-Bit using GCC 9.2.1
- FreeBSD 12.1 64-Bit using clang 8.0.1
- Windows 10 64-Bit, GCC 9.2 (MinGW-w64 provided by MSYS2)

by following the following procedure:
```
git clone https://github.com/simulton/qschematic
cd qschematic
mkdir .build
cd .build
cmake ..    # (optionally setting CMAKE_CXX_COMPILER to `gcc` or `clang`)
make
sudo make install
make qschematic-demo
```
This assumes that GPDS has been installed successfully. However, the errors you're showing do not seem to be related and looking at #16 I assume that that wouldn't be a source of problems anyway.

What version of Qt are you using? The QSchematic library has been used successfully with Qt 5.12 and Qt 5.13. I do believe that we also used this library with Qt 5.11 libraries but I'm unsure about that.

ckqee, 2020-03-17:
This is good information! I am currently on Qt5.9.5 

```
qmake -version
QMake version 3.1
Using Qt version 5.9.5 in /usr/lib/x86_64-linux-gnu
```

Before I break my software packages which rely on this particular version - I will do a backup and get back to you.
Noticed that there is no Qt5.13 version available for download
from:
https://download.qt.io/snapshots/qt/

```

5.15/ | 12-Feb-2020 14:43 | - |  
-- | -- | -- | --
5.14/ | 12-Feb-2020 14:43 | - |  
5.12/ | 12-Feb-2020 14:00 | -

```

Tectu, 2020-03-23:
Can't help you there - we use the distributions provided by the Qt installer.

In any case I have no idea what would lead to the error you're showing. I've checked our CI and I've  done manual builds of the `develop` branch myself and I got it to run on Windows, Linux & BSD with GCC and clang out of the box.

Looking forward hearing what the issue boiled down to.

Tectu, 2020-03-30:
In the meantime we've improved the build system. The cmake scripts will now automatically pull in the GPDS dependency and everything will build out of the box after cloning the repository.
The `DEPENDENCY_GPDS_FROM_SYSTEM` variable can be set to `ON` to include the GPDS dependency from the system using `find_package()` instead.

We've also added a badge to the readme indicating the state of the continuous integration builds.

ckqee, 2020-05-10:
Hi,
Sorry for the delay. Hope all is well with you considering the global situation.

I am stuck at the same place in getting QSchematic to compile:

```
user1@machine:~/github/simulton/QSchematic$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

nothing to commit, working tree clean
user1@machine:~/github/simulton/QSchematic$ git pull
Already up-to-date.
user1@machine:~/github/simulton/QSchematic$ mkdir .build
user1@machine:~/github/simulton/QSchematic$ cd !$
cd .build
user1@machine:~/github/simulton/QSchematic/.build$ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/.build/_deps/gpds-build/test/doctest
Scanning dependencies of target doctest
[ 12%] Creating directories for 'doctest'
[ 25%] Performing download step (git clone) for 'doctest'
-- doctest download command succeeded.  See also /home/user1/github/simulton/QSchematic/.build/_deps/gpds-build/test/doctest/doctest-prefix/src/doctest-stamp/doctest-download-*.log
[ 37%] No patch step for 'doctest'
[ 50%] No update step for 'doctest'
[ 62%] No configure step for 'doctest'
[ 75%] No build step for 'doctest'
[ 87%] No install step for 'doctest'
[100%] Completed 'doctest'
[100%] Built target doctest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/.build
user1@machine:~/github/simulton/QSchematic/.build$ 
user1@machine:~/github/simulton/QSchematic/.build$ make
Scanning dependencies of target wire_system_autogen
[  2%] Automatic MOC for target wire_system
[  2%] Built target wire_system_autogen
Scanning dependencies of target wire_system
[  4%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/wire_system_autogen/mocs_compilation.cpp.o
In file included from /home/user1/github/simulton/QSchematic/.build/qschematic/wire_system/wire_system_autogen/EWIEGA46WW/moc_manager.cpp:9,
                 from /home/user1/github/simulton/QSchematic/.build/qschematic/wire_system/wire_system_autogen/mocs_compilation.cpp:2:
/home/user1/github/simulton/QSchematic/qschematic/wire_system/manager.h:26:32: error: ISO C++ forbids declaration of ‘Q_DISABLE_COPY_MOVE’ with no type [-fpermissive]
   26 |     Q_DISABLE_COPY_MOVE(manager)
      |                                ^
/home/user1/github/simulton/QSchematic/qschematic/wire_system/manager.h:26:32: error: expected ‘;’ at end of member declaration
   26 |     Q_DISABLE_COPY_MOVE(manager)
      |                                ^
      |                                 ;
qschematic/wire_system/CMakeFiles/wire_system.dir/build.make:62: recipe for target 'qschematic/wire_system/CMakeFiles/wire_system.dir/wire_system_autogen/mocs_compilation.cpp.o' failed
make[2]: *** [qschematic/wire_system/CMakeFiles/wire_system.dir/wire_system_autogen/mocs_compilation.cpp.o] Error 1
CMakeFiles/Makefile2:1320: recipe for target 'qschematic/wire_system/CMakeFiles/wire_system.dir/all' failed
make[1]: *** [qschematic/wire_system/CMakeFiles/wire_system.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
user1@machine:~/github/simulton/QSchematic/.build$ 
```

ckqee, 2020-05-10:
Currently on:
```
uname -a
Linux machine 4.15.0-96-generic #97-Ubuntu SMP Wed Apr 1 03:25:46 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

Tectu, 2020-05-27:
Sorry for the long silence...

Checkout the latest `develop` branch. You should be able to build 80c47f2c260d3ba6843b6248d2168c7ecd38382c.

ckqee, 2020-06-03:
Yet so close but too far

```
user1@machine:~/github/simulton/QSchematic$ git checkout develop
Branch 'develop' set up to track remote branch 'develop' from 'origin'.
Switched to a new branch 'develop'
user1@machine:~/github/simulton/QSchematic$ ll
total 44
drwxr-xr-x  7 user1 user1 4096 Jun  3 16:13 .
drwxr-xr-x 20 user1 user1 4096 Jun  3 16:11 ..
drwxr-xr-x  6 user1 user1 4096 Jun  3 16:14 .build
-rw-r--r--  1 user1 user1  937 Jun  3 16:10 CMakeLists.txt
drwxr-xr-x  8 user1 user1 4096 Jun  3 16:10 demo
drwxr-xr-x  5 user1 user1 4096 Jun  3 16:10 docs
drwxr-xr-x  8 user1 user1 4096 Jun  3 16:16 .git
-rw-r--r--  1 user1 user1  126 Jun  3 16:10 .gitignore
-rw-r--r--  1 user1 user1 1085 Jun  3 16:10 license.txt
drwxr-xr-x  6 user1 user1 4096 Jun  3 16:10 qschematic
-rw-r--r--  1 user1 user1 3319 Jun  3 16:10 readme.md
user1@machine:~/github/simulton/QSchematic$ cd .build/
user1@machine:~/github/simulton/QSchematic/.build$ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/.build/_deps/gpds-build/test/doctest
[100%] Built target doctest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/.build
user1@machine:~/github/simulton/QSchematic/.build$ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/.build/_deps/gpds-build/test/doctest
[100%] Built target doctest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user1/github/simulton/QSchematic/.build
user1@machine:~/github/simulton/QSchematic/.build$ make
[  2%] Automatic MOC for target wire_system
[  2%] Built target wire_system_autogen
Scanning dependencies of target wire_system
[  4%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/wire_system_autogen/mocs_compilation.cpp.o
[  6%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/line.cpp.o
[  9%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/manager.cpp.o
[ 11%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/wire.cpp.o
[ 13%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/point.cpp.o
[ 15%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/net.cpp.o
[ 18%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/__/utils.cpp.o
[ 20%] Building CXX object qschematic/wire_system/CMakeFiles/wire_system.dir/__/settings.cpp.o
[ 20%] Built target wire_system
Scanning dependencies of target gpds-objs
[ 22%] Building CXX object _deps/gpds-build/lib/CMakeFiles/gpds-objs.dir/src/archiver_xml.cpp.o
[ 25%] Building CXX object _deps/gpds-build/lib/CMakeFiles/gpds-objs.dir/src/container.cpp.o
[ 27%] Building CXX object _deps/gpds-build/lib/CMakeFiles/gpds-objs.dir/src/value.cpp.o
[ 29%] Building CXX object _deps/gpds-build/lib/CMakeFiles/gpds-objs.dir/src/attributes.cpp.o
[ 31%] Building CXX object _deps/gpds-build/lib/CMakeFiles/gpds-objs.dir/src/serialize.cpp.o
[ 34%] Building CXX object _deps/gpds-build/lib/CMakeFiles/gpds-objs.dir/3rdparty/tinyxml2/tinyxml2.cpp.o
[ 34%] Built target gpds-objs
Scanning dependencies of target gpds-static
[ 36%] Linking CXX static library libgpds.a
[ 36%] Built target gpds-static
Scanning dependencies of target qschematic-objs_autogen
[ 38%] Automatic MOC and UIC for target qschematic-objs
[ 38%] Built target qschematic-objs_autogen
Scanning dependencies of target qschematic-objs
[ 40%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/qschematic-objs_autogen/mocs_compilation.cpp.o
[ 43%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commandbase.cpp.o
[ 45%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemadd.cpp.o
[ 47%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemmove.cpp.o
[ 50%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemremove.cpp.o
[ 52%] Building CXX object qschematic/CMakeFiles/qschematic-objs.dir/commands/commanditemvisibility.cpp.o
[... 134 lines cut ...]
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewer.h:20:40: error: expected ‘;’ at end of member declaration
   20 |     Q_DISABLE_COPY_MOVE( NetlistViewer )
      |                                        ^
      |                                         ;
In file included from /home/user1/github/simulton/QSchematic/.build/demo/qschematic-demo_autogen/DRL6IFWHKL/moc_netlistviewerview.cpp:9,
                 from /home/user1/github/simulton/QSchematic/.build/demo/qschematic-demo_autogen/mocs_compilation.cpp:15:
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewerview.h:13:44: error: ISO C++ forbids declaration of ‘Q_DISABLE_COPY_MOVE’ with no type [-fpermissive]
   13 |     Q_DISABLE_COPY_MOVE( NetlistViewerView )
      |                                            ^
/home/user1/github/simulton/QSchematic/demo/netlistviewer/netlistviewerview.h:13:44: error: expected ‘;’ at end of member declaration
   13 |     Q_DISABLE_COPY_MOVE( NetlistViewerView )
      |                                            ^
      |                                             ;
[... cut ...]
make[3]: *** [demo/CMakeFiles/qschematic-demo.dir/qschematic-demo_autogen/mocs_compilation.cpp.o] Error 1
CMakeFiles/Makefile2:1402: recipe for target 'demo/CMakeFiles/qschematic-demo.dir/all' failed
make[2]: *** [demo/CMakeFiles/qschematic-demo.dir/all] Error 2
CMakeFiles/Makefile2:1409: recipe for target 'demo/CMakeFiles/qschematic-demo.dir/rule' failed
make[1]: *** [demo/CMakeFiles/qschematic-demo.dir/rule] Error 2
Makefile:697: recipe for target 'qschematic-demo' failed
make: *** [qschematic-demo] Error 2

```

Tectu, 2021-03-01:
Any news on this? The cmake scripts have been heavily improved in the meantime.
This library now builds properly on MacOS, Ubuntu, FreeBSD and Windows with GCC and Clang.

closed 2021-05-14 by Tectu

Vineg, 2022-01-31:
I had a similar problem with the build.
I've turned on build debug output and found that there was older development version of Qt library installed with apt-get.
`sudo apt-get remove qtbase5-dev` solved this issue for me.

serg-z, 2022-09-06:
FYI `Q_DISABLE_COPY_MOVE` was introduced in Qt 5.13, so to build the library you'll need at least 5.13.

Tectu, 2022-09-07:
@serg-z Technically, when using CMake, Qt5.15 is needed as the `readme.md` reflects:
```
Dependencies:
   - Qt5 (>= 5.15) or Qt6
   - GPDS for (de)serialization.
```
Although this obstacle could be overcome by manually handling the CMake target version (Qt 5.15 introduced version-less targets).