simulton/QSchematic #19: VS2019 build...

RallyTronics, 2020-04-12:
I have been struggling to get this to build in VS2019 but I now have the libraries building.  If anyone is trying to build this in VS2019 you have to:

- add the Za compiler option as VS does not support the _not, and, and or_ keywords natively.
- remove the Wextra flag, VS doesn't understand it
- add #include <optional> in a few files

 My next challenge is that although I have the BUILD_DEMO selected it doesn't create the project and add it to my visual studio solution.  Any thoughts?  I am not much of a cmake guy.

Tectu, 2020-04-15:
Thank you for sharing your findings. We'll try to update the documentation (readme) and cmake build scripts to make this easier. Suggestions & help are of course very welcome - we have very limited experience working with MSVC and VS.

> add #include optional in a few files
Could you be a bit more precise here? While we never tried to compile QSchematic using MSVC we definitely used several different compilers. The only reason why I would see this happening is if the standard library shipped with MSVC has a different include hierarchy within the implementation.
If you could provide us with a bit more information here we'd be glad to add the missing includes to ensure that compilation with MSVC will run smoothly.

> My next challenge is that although I have the BUILD_DEMO selected it doesn't create the project and add it to my visual studio solution. Any thoughts? I am not much of a cmake guy.
Unfortunately, I'm not much of a MSVC guy :p
As far as I can remember setting `BUILD_DEMO` to `ON` will only generate the necessary build system files (project files for VS?) but not actually issue the build itself. If you set `BUILD_DEMO` to `ON` you should see the `qschematic-demo` target which you can select and build.
I don't know how this works with VC. When working with GCC on Windows it's just a matter of doing this:
```
cd /path/to/qschematic
mkdir .build
cd .build
cmake .. -DBUILD_DEMO=ON
make qschematic-demo
```
I hope this helps.

RallyTronics, 2020-04-15:
Here are more detailed steps

- remove -Wextra
- remove -g
- add -Za to handle the not,or,and ... keywords
- add `#include<stdexcept>` to value.cpp
- replace GPDS_ASSERT() with assert() as VS2019 doesn't know what __PRETTY_FUNCTION__ is
- add `#include <optional>` to manager.h

I get a number of unresolved external symbol errors with qschematic-shared but i haven't looked into those as I am interesting in the static lib right now.

Tectu, 2020-04-16:
Most of these appear to be related to [GPDS](https://github.com/simulton/GPDS). I'll try to get to that soon.

Were you able to build & use the static library (and demo application)?

RallyTronics, 2020-04-16:
Not yet, the demo project compiles after fixing some include but is having problems linking with Qt for some reason.  Haven't had time to get back to it.

traits, 2020-05-27:
Have the same problems here. Would it be possible reporting issues for the GPDS repo himself? The final software looks promising, so I would spend some work into making the library(ies) more platform-agnostic in regard to Windows. It would be a pity if it was ignored by users of the OS just for build system blocking issues and few hard-wired gcc peculiarities.

Tectu, 2020-05-27:
Any kind of contribution is definitely highly welcomed. This library was created with the goal to provide a ready-to-use solution to draw diagrams & schematics that is easy to integrate. I only have a few months of experience working with cmake so there's definitely a lot of room for improvement that I'm very open to.

We're successfully building & running this QSchematic library on Windows, Ubuntu & FreeBSD so far.

traits, 2020-05-27:
> We're successfully building & running this QSchematic library on Windows, Ubuntu & FreeBSD so far.

Obviously on Windows the project utilizes some msys/mingw-based toolchain. I think, native MSVC support would be worth the work. For many developers on this OS it is still the gold standard (more to the point, some technical reasons forbid or at least make it harder mixing mingw and MS-based software artefacts).

Tectu, 2020-05-27:
I fully agree. I only did a few projects with MSVC myself but I'm aware of the fact that it's the standard on Windows platforms.
Pull requests are highly welcomed - I'm looking forward improving the build scripts & overall portability of the library!

traits, 2020-05-28:
Ok, I made a fork and will get back to you (no hard schedule for now).

Another thing to project structure:
There are probably reasons, but listening to my gut feelings the build architecture could be possibly simplified regarding submodules etc.. 
At first glance, the tinyxml projects each consist of only two or three files. Maybe its better to include a defined commit/release whatever of these files as deep copy in GPDS and updating them in a controlled manner from time to time. No final opinion about the relationship GPDS/QSchematic in this regard, though. 
In general, simple drop-ins are often attractive to application developers. A low number of sources qualifies most of the time for this, IMO.

Tectu, 2020-06-01:
I think you're right about the submodules. The TinyXML2 and TinyXML2-ex submodules have now been removed from the GPDS repository. The corresponding sources have been added as hard-copies to the repository.
There was a pull request regarding the exception in the meantime. However, the TinyXML2-ex part needs a proper rewrite anyway. It comes with a lot of compiler warnings.

traits, 2020-06-08:
I think, I have to stop here. Finally, I haven't the time to weed out all the OS-specific issues, but will at least give some rudimentary feedback. 

Summary:

- got GPDS to work
- QSchematic misses some/all exports (wire, net, probably more) on Windows and the dll will not link because of this. 

The cmake stuff can probably be fixed, but seems in its current shape overly complex (in relation to the softwares scope and size) to me. It consumes simply too much of my time to understand all the ramifications. My advice would be to simplify things to the bare essentials build-system and project-structure-wise. Either by merging the two libraries, clearly state unix/mingw-only development or active testing in a native Windows environment.

BriFuture, 2020-06-15:
Hi, have you ever hit this error information:

[image]

I followed steps from @RallyTronics , but this error confused me as xmemory0 comes with VS2017.

Tectu, 2020-06-19:
Unfortunately I do not have access to an MSVC based environment. Maybe somebody else can give this a go. For now I'll update the documentation to include information about tested environments & compilers as per @traits suggestions.

One thing tho, regarding @BriFuture : QSchematic requires a C++17 compatible compiler. I have very little to no experience working with MSVC. Is VS2017 C++17 compatible?

BriFuture, 2020-06-21:
VS2017 is  C++17 compatible.

BriFuture, 2020-06-22:
The problem `xmemory0` seems to be related with `gpds::container` add_value
[image]

as the screenshort shows, the C2665 error is eliminated if all these `add_value` call were commented. But more errors came up:

[image]

By the way, the gpds project will be successfully compiled sololy.

Tectu, 2021-05-16:
@chehrlic Seems to be able to compile the library using MSVC2019 after some patch which has been merged into the `master` branch.

closed 2021-05-16 by Tectu