simulton/QSchematic #20: Mixing Branches

traits, 2020-05-31:
https://github.com/simulton/QSchematic/blob/80c47f2c260d3ba6843b6248d2168c7ecd38382c/qschematic/external.cmake#L10

I had a few problems with this. I think, QSchematic/master should always refence GPDS/master for coherence and independence from current development.

Tectu, 2020-05-31:
I assume that therefore this would need to be a conditional settings: If QSchematic is on the `master` branch, use GPDS `master` branch; if QSchematic is on the `develop` branch, use the GPDS `develop` branch.
How would you handle this through cmake?

Ultimately QSchematic should never use the `develop` branch of GPDS as GPDS is supposed to be a standalone library with its own development & release cycle.

traits, 2020-06-02:
Personally, I'm avoiding downloading or even getting information for repositories from cmake in general, because it mixes source control (git) with build (cmake) and sometimes even deployment responsibilities. 

So, I would advocate for a submodule or even let the user build/install things on his own, only specifying the required interface without putting too much emphasis on a technical solution for everything.

Tectu, 2020-06-02:
Well, what you're explaining is exactly where I was coming from :p
This library (together with GPDS) is the first time I ever used cmake. It was hard for me to judge how things should be handled "ideally".

At the beginning I just relied on people building & installing GPDS separately. However, that seemed cumbersome for most people that are just used to downloading a Qt project and clicking the green arrow in Qt creator. Furthermore, setting up CI seemed a lot easier when not having to handle dependencies externally too.
Therefore, my solution is/was to have a cmake option that allows the user to either use the GPDS library already on the system or to pull & build it internally. This way both scenarios should be covered.

Tectu, 2021-03-01:
The way that the GPDS dependency is handled has been vastly improved.
There's now the option `DEPENDENCY_GPDS_DOWNLOAD` which is defaulted to `ON`. If enabled, CMake's `FetchContent` module will be used to pull the GPDS repository automatically during compile time.

Furthermore, `DEPENDENCY_GPDS_TARGET` can be set to which-ever GPDS target one wants to link against. It defaults to `gpds-shared`. Setting this to `gpds-static` easily allows to build it as a static dependency instead.

closed 2021-03-01 by Tectu