simulton/QSchematic #21: Building the demo from Qt Creator?

ladmanj, 2020-12-19:
Dear authors

I know this isn't really a bug report, but I'm continuously suffering with the fact, that there on github isn't any "free discussion" section within the repositories.

I have successfully built gpds,  QSchematic and also the demo using cmake.
The demo works as expected.

Because I'm mostly hardware engineering oriented developer, I'm not really familiar with CMake, the modern concepts of latest C++ standards and so on, but I have already created many simple tools using Qt.

I have developed a hardware module with a microcontroller, which is configurable over Modbus network, you can imagine it as simplified version of PLC or configurable relay. I have created also small and simple tool using Qt, for debugging the Modbus communication.

Now I want to merge it with the QSchematic demo code, to see the internal connections and settings as a schematic.

And the problem is, that if I try to open and build the demo in Qt Creator (pointing it to CMakeList.txt), the project is imported but can't be built.

I am working on kubuntu 20.10 with Qt Creator and tools, open source version installed by official Qt installer and everything is up to date.

The include directories aren't set properly somehow
```
[ 13%] Building CXX object CMakeFiles/qschematic-demo.dir/qschematic-demo_autogen/mocs_compilation.o
In file included from /usr/local/include/qschematic/items/wireroundedcorners.h:3,
                 from /home/ladmanj/DataDisk/src/demo/items/fancywire.h:3,
                 from /home/ladmanj/DataDisk/src/build-demo-Desktop_Qt_5_7_1_GCC_64bit-Debug/qschematic-demo_autogen/HU7TA6T7T7/moc_fancywire.cpp:10,
                 from /home/ladmanj/DataDisk/src/build-demo-Desktop_Qt_5_7_1_GCC_64bit-Debug/qschematic-demo_autogen/mocs_compilation.cpp:3:
/usr/local/include/qschematic/items/wire.h:5:10: fatal error: wire_system/point.h: No such file or directory
    5 | #include "wire_system/point.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Thank you for any help.

Best regards
Jakub

ladmanj, 2020-12-20:
Hello
It seems I have managed to compile the demo (without CMake, but this is probably irrelevant).
The problem was, the qt creator bundled compiler doesn't fully support c++17, as the system one does.
I have only qt-5.7 installed in qt creator, but the system one is qt-5.14.2.

I have added /usr/bin/qmake as base for new kit and library and now the demo is compiled.

Regards

Tectu, 2020-12-21:
Hi,

I'm a hardware developer too so I feel your "pain" there - no worries :)

Just to be sure: You managed to get everything up and running the way you need it now?
It is correct that you need a C++17 compatible compiler. Which one you installed along with Qt-Creator depends on the system and packages you're using.

Don't hesitate to ask if there are any other questions. An issue is not necessarily a bug anyway.

ladmanj, 2020-12-29:
I'm not done with this project by any means, but it seems this issue is ready to be closed. So I'm doing that.
Thanks

closed 2020-12-29 by ladmanj