simulton/QSchematic #23: Licensing

LeeRuns, 2021-04-25:
Not to be dense, but this is built on top of qt right? So can it be mit licensed?

Feel free to educate me. As some background i am working on a proprietary code base, never used qt. Have toyed with the idea lately given the schamatica library appears to be only thing close to what I need

Tectu, 2021-04-26:
This library does not ship with Qt. It's simply a library that internally uses/relies on Qt. This library's license does not affect the Qt license (eg. using this library does not make Qt suddenly become MIT licensed itself).

This library is MIT licensed and can therefore be used with Qt's GPL or LGPL license (or the commercial once).

As usual: **I am not a lawyer.**

closed 2021-05-07 by Tectu

LeeRuns, 2021-07-13:
Just realized i had not responded. Thanks for the clarification on this. Opensource can be tricky this way. Had to do a deeper dive into licensing for another project. 100% agree with the feedback. This was learning opportunity for me. Thanks for leading

Tectu, 2021-07-13:
Thanks for reporting back :)
Just make sure that you don't rely on any legal information provided by me.