simulton/QSchematic #24: error: libqschematic.so with undefined references

bond0O7, 2021-05-07:
Hi there :)
I've built the library and tried to link it with my qt application, I'm using qmake to build my project
I've added LIBS += -lgpds -lqschematic to my .pro file
When I compile the project it gives me tons of linker errors of undefined references
Here are few:
/usr/bin/ld: /usr/local/lib/libqschematic.so: undefined reference to `typeinfo for QSchematic::Item'
/usr/bin/ld: /usr/local/lib/libqschematic.so: undefined reference to `QSchematic::Node::sizeChanged()'
/usr/bin/ld: /usr/local/lib/libqschematic.so: undefined reference to `vtable for QSchematic::Connector'
/usr/bin/ld: /usr/local/lib/libqschematic.so: undefined reference to `QSchematic::Scene::itemRemoved(std::shared_ptr<QSchematic::Item const>)'
.
.

Am I missing something
I've built the library with the instructions provided from the README.me file with **DEPENDENCY_GPDS_DOWNLOAD** enabled and extra **_sudo make install_** to install it on my system

Thanks in advance

bond0O7, 2021-05-07:
Okay
I made a small project with cmake, part of the CMakeLists.txt is:

```
find_package(QT NAMES Qt5 COMPONENTS Widgets REQUIRED)
find_package(Qt${QT_VERSION_MAJOR} COMPONENTS Widgets REQUIRED)
find_package(gpds REQUIRED)
find_package(QSchematic REQUIRED)

set(PROJECT_SOURCES
        main.cpp
        MainWindow.cpp
        MainWindow.hpp
        MainWindow.ui
)
add_executable(testWitCmake
    ${PROJECT_SOURCES}
)

target_link_libraries(testWitCmake PRIVATE Qt${QT_VERSION_MAJOR}::Widgets qschematic::qschematic-shared)

```

The same undefined references errors occurred

Then I checked the symbols of libqschematic.so with **_nm -u | c++filt_** to check the undefined symbols(It's a long list) but what I've noticed is that it has some methods from the standard Qt included like a lot of _QGraphicsItem, QGraphicsView...etc_

```
U QSchematic::ItemMimeData::staticMetaObject
U QSchematic::Item::movedInScene(QSchematic::Item&)
U QSchematic::Item::settingsChanged()
U QSchematic::Item::highlightChanged(QSchematic::Item const&, bool)
U QSchematic::Item::staticMetaObject
U QSchematic::Item::moved(QSchematic::Item&, QVector2D const&)
U QSchematic::Item::rotated(QSchematic::Item&, double)
U QSchematic::Node::sizeChanged()
U QSchematic::View::modeChanged(QSchematic::View::Mode)

U QGraphicsItem::clearFocus()@Qt_5
U QGraphicsItem::itemChange(QGraphicsItem::GraphicsItemChange, QVariant const&)@Qt_5
U QGraphicsItem::setVisible(bool)@Qt_5
U QGraphicsItem::setRotation(double)@Qt_5
U QGraphicsItem::setSelected(bool)@Qt_5
U QGraphicsItem::unsetCursor()@Qt_5

and more
```

Tectu, 2021-05-11:
Which compiler are you using?

I am using this library successfully in two different applications. Albeit both of these applications are cmake based and use GCC 10.

Tectu, 2021-05-13:
I think I managed to reproduce the problem - will continue investigating.

Tectu, 2021-05-14:
Could you have a look at the current `master` branch and inspect the built shared object again?
Also: It would be wonderful if you could test the static library as well.

Tectu, 2021-05-16:
I've put some work into this. It turns out that once should not build the shared & static library variants from cmake's object library.
Please checkout release `1.0.2` and report back!

Tectu, 2021-06-14:
I'm going to close this due to inactivity.
Release `1.1.0` is known to have fixed this issue and also builds out of the box on MSVC.

closed 2021-06-14 by Tectu