simulton/QSchematic #29: Moving a point of a wire, from graphics does not respect straight angles

ChristianWieden, 2021-06-12:
Moving a wire point via UI calls mouseMoveEvent in items/wire.cpp which then hands over the move to class CommandWirepointMove.
This class then does the transformation in the Constructor, but it does not check if straight angles should be preservered.

But wire_system/wire.cpp has this functionality, and has testcases for this, so why not call this function?

Tectu, 2021-06-12:
I think this is a simple oversight from when the dedicated/separated wire system was introduced.
Before this the wire handling algorithms were baked into the overall code of the library. The idea of separating the wire system into a separate component was to make the code more portable and easier to test without anything being tied to the UI side of things.

This new component certainly needs some polishing and it's likely that functionality was not migrated completely - Pull requests are welcomed :)

Tectu, 2025-06-11:
Commit d1b4bdc26c39255c212f0603d38786738a20aaef cleans up the implementation of `Commands::WirepointMove` by using the logic implemented in the wiresystem.

The actual problem reported in this issue (not respecting the straight angles setting) is actually on purpose.
If the user moves just a point, this is the expected behavior.
If the user wants to move a line segment (i.e. two points), they can instead grab and drag (move) the wire segment which works as expected.

closed 2025-06-11 by Tectu