simulton/QSchematic #38: why I can't open this project with Qt Creator(Qt6.3.2)?

isteelcx, 2022-11-22:


Tectu, 2022-11-22:
Can you please provide a bit more details such as what "can't open means"? Do you get an error message?

isteelcx, 2022-11-23:
here is Qt6.3.2 open error :
[image]
why can't Qt5.14.2 open neither?
[image]

isteelcx, 2022-11-23:
and can you update the way to open this project by the root CMakeLists.txt?
[image]

isteelcx, 2022-11-24:
thank you for support, now can be open by Qt6 and build pass

isteelcx, 2022-11-24:
1. Can you talk about your understanding of (paceholder/nodeeditor)?

2. What do you think are the advantages and disadvantages of this project? First of all, I think its advantages are the realization of node based editor based on Model-View design pattern, with strong customization and extensibility. As for its disadvantages, I think its code writing style is not good, and the writing is not beautiful enough. Strange indentation and naming affect the aesthetic feeling. And there is no support for projects such as thedmd/imgui-node-editor where a small node can be included in a large node.

3. What features do you think a fully scalable node based editor architecture should meet?

isteelcx, 2022-11-25:
Another question, do you follow the official Qt coding specification when you write your code?
https://wiki.qt.io/Qt_Coding_Style

Tectu, 2022-11-29:
Sorry for the late reply. I took some time to figure out how to respond to your question. I came to the conclusion that the best I can tell you is the universal advice in a situation like this: Evaluate and pick yourself. I cannot possible provide you with any objective, 100% non-biased opinion/answer. Furthermore, some of the reasons which resulted in me starting to write `QSchematic` all those years ago were required features which were missing in `paceholder/nodeeditor`. Most likely some, if not all of those features have been added by now. After all, it has been almost eight years.
One thing that can be called an objective assertion: `QSchematic` is backed by a company, not just a single individual.

> Another question, do you follow the official Qt coding specification when you write your code?

No, we don't. However, coding styles are usually a personal preference. Most often only novices and religious people care about _which_ coding style is being used while everybody else just cares about _consistency_.
Unfortunately, the `QSchematic` code base coding style is not consistent as of today. That is one of the things we will rectify with the 2.0 release.

isteelcx, 2022-11-30:
will QSchematic support Group Nodes and connet line multiple styles，just like：
[image]

[image]

and changeable nodes shapes：
[image]

isteelcx, 2022-11-30:
@Tectu After a period of experience,this is realy a great project

Tectu, 2022-12-01:
As of today, QSchematic provides full visual customization to the user ("styles" and "shapes"). You can re-implement the `::pain()` function of each item. This is actually demonstrated in the supplied demo. A custom style is implemented (the gray items with the drop shadow) and custom shapes are demonstrated with the "Flow Start" and "Flow End" items.

Group nodes are implicitly supported. i.e. what you showed in your screenshot works with QSchematic. However, the built-in netlist generator currently doesn't support nested/grouped nodes. This is something that is on our short-term ToDo list as we need that feature ourselves in a product :)

isteelcx, 2022-12-06:
Will you change the basic architecture of QSchematic in the future?

Tectu, 2022-12-06:
What kind of questions are these?
Will you change your kidney in the future? Probably not, right? But you still wouldn't promise it (I assume). :D

isteelcx, 2022-12-07:
I hope you can understand my concern about these particularities. I prefer software architecture issues to functional detail changes, because this affects the stability of the development model in the long run. As I was worried earlier, maybe at some point you suddenly have a more sophisticated architectural design concept and overturn the current project. I hope you can understand my concern about these particularities. I prefer software architecture issues to functional detail changes, because this affects the stability of the development model in the long run. As I worried earlier, At some point you may have a more sophisticated architectural design concept that overturns the current project, and it will be awkward to be the user of this awesome project

Tectu, 2022-12-07:
There are currently no plans to go down that road. If that would happen, it would simply be a new major version the same way that most software is being developed & distributed :)

isteelcx, 2022-12-10:
Could you share your feature update schedule? Such as the node grouping mentioned earlier

Tectu, 2023-04-27:
Closing this due to inactivity.

closed 2023-04-27 by Tectu