simulton/QSchematic #42: Connect after rotation

tprohor, 2023-08-07:
In demo application: after demo1 figure rotation to some angle it is impossible to connect wire to figure

Tectu, 2023-08-07:
Hi,

Thanks for reporting this. Could you add a small video / screencap illustrating the issue just to be sure that we're talking the same thing? Something similar to #33.

tprohor, 2023-08-10:
https://github.com/simulton/QSchematic/assets/29174324/77dde9fa-c0cc-4176-9b50-159f3f6a062d

tprohor, 2023-08-10:
And another problem: demo crash at close event
[image]

Tectu, 2023-08-10:
The crash on close has been fixed in 2045bf291fdd56f4cf8a3f98d08e2aef7770909a - thanks!

Regarding the connection issue: This is a side-effect of the snap-to-grid policy. I think the best course of action is to add a "magnetic" feature: When drawing a wire, getting close to a connector will lead to a successful connection. Currently, wire connection points are restricted to the grid.