simulton/QSchematic #50: ninja: error: CMakeFiles/impl-Debug.ninja:675: multiple rules generate qschematic/Debug/qschematicd.lib

aboelens, 2023-11-21:
Building QSchematic on windows 10 I get the error message:

ninja: error: CMakeFiles/impl-Debug.ninja:675: multiple rules generate qschematic/Debug/qschematicd.lib

This is due to the static and shared library having the same name. Either removing the shared or static target from the CMakeList.txt gets rid of the error message.

Tectu, 2023-11-21:
Quick question: Is that using Visual Studio?

aboelens, 2023-11-21:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> External email

Hi Joel,

That is building from PowerShell.

Met vriendelijke groet / Kind regards,
--
Arnout Boelens
Principal Researcher Thermal Processes
R&D - Rolling, Finishing & Measurement
T: +31 (0) 251 495 136

Sensitivity: general

This transmission is confidential and must not be used or disclosed by anyone other than the intended recipient. Neither Tata Steel Europe Limited nor any of its subsidiaries can accept any responsibility for any use or misuse of the transmission by anyone.

For address and company registration details of certain entities within the Tata Steel Europe group of companies, please visit https://www.tatasteeleurope.com/en/legal-notice/entities

**********************************************************************

Tectu, 2023-11-22:
This problem has been popping up on the Windows side before. I'll get that fixed but that might take a couple of days (definitely not before the weekend).

I suggest that you remove the corresponding CMake `add_library()` call to either the static or the shared library until we've fixed this in the upstream repository.

aboelens, 2023-11-23:
Thanks! Will do

Tectu, 2023-11-28:
This should now be fixed in the current master branch. Can you please check?

Note that you'll have to manually supply the correspondingly updated version of GPDS which also suffered from this problem.
See https://github.com/simulton/gpds/issues/16

closed 2023-12-04 by Tectu