simulton/QSchematic #51: _deps\gpds-src\lib\include\gpds\utils.hpp(41): error C2039: 'to_string': is not a member of 'std'

aboelens, 2023-11-21:
Building version 1.5.0 with MSVC on Windows 10 I get the error message

_deps\gpds-src\lib\include\gpds\utils.hpp(41): error C2039: 'to_string': is not a member of 'std'

Adding "#include <string>" to utils.hpp fixes the issue.

Tectu, 2023-11-21:
That appears to be an issue related to GPDS, not QSchematic: https://github.com/simulton/gpds

aboelens, 2023-11-21:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> External email

Thanks. I’ll report it there.

Met vriendelijke groet / Kind regards,
--
Arnout Boelens
Principal Researcher Thermal Processes
R&D - Rolling, Finishing & Measurement
T: +31 (0) 251 495 136

Sensitivity: general

This transmission is confidential and must not be used or disclosed by anyone other than the intended recipient. Neither Tata Steel Europe Limited nor any of its subsidiaries can accept any responsibility for any use or misuse of the transmission by anyone.

For address and company registration details of certain entities within the Tata Steel Europe group of companies, please visit https://www.tatasteeleurope.com/en/legal-notice/entities

**********************************************************************

Tectu, 2023-11-28:
See https://github.com/simulton/gpds/issues/16