simulton/QSchematic #52: how can i use this lib in my pyside6 project?

PeaceZhang, 2023-11-24:
Hello, Simulton Experts,

First thank you for you did this great job.
I think this library is very suitable for autosar swc architecture design, and now i have a idea to develop a GUI for my opensource project:https://github.com/PeaceZhang/ArxmlTools. for the GUI solution, I want use Pyside6 to develop it. and i want integrate QSchematic to it.

So do you have any solution for this topic?

Best regards,
heping,zhang

Tectu, 2023-11-25:
Hi,

There are plenty of libraries written in C++ with a corresponding Python wrapper so you can use it from Python. Qt itself is a good example.
Therefore, we're not aware of any reason as to why you couldn't use this QSchematic library with Python.

You'll have to write the Python wrapper but there are plenty of documentation, tutorials and examples out there on that subject.

bjornstromberg, 2024-02-01:
i have not myself digged deep into it, but start digging from shiboken side https://doc.qt.io/qtforpython-6/shiboken6/index.html

no clue how much work it will be to generate the python interfaces but it should be possible as far as i have understood.

closed 2024-05-11 by Tectu

svanimisetti, 2024-12-02:
[Python-SIP](https://github.com/Python-SIP/sip) was created for this very purpose - the PyQt binding are generated using SIP. I am interested in using QSchematic with PyQt5, and will attempt a test integration. If I am successful, I will post my findings in separate issue.