simulton/QSchematic #56: ItemsCustodian defective due to logical address error of 'Item'

bjornstromberg, 2024-01-31:
added all headers of 1.5.0 just to testbuild all the templates, and it seems that ItemsCustodian has not updated the logical adress when `QSchematic::Item` got moved into `QSchematic::Items::Item` a while back...

`if constexpr (std::is_base_of_v<QSchematic::Item, CustodyItemT>)` should be `if constexpr (std::is_base_of_v<QSchematic::Items::Item, CustodyItemT>)`

this issue is still in master @ 52416bcd17ca2bd65ccbff4f46882897454d652d

Tectu, 2024-01-31:
Good find, thanks! That custodian is currently not used but you are entirely correct.

I can throw out a `1.5.2` release if that helps your packaging efforts?

closed 2024-01-31 by Tectu

bjornstromberg, 2024-02-01:
yes please a 1.5.2 would be nice.

thanks for fast fixes. i've fleshed out simulton/gpds#18 to work and attached patches so only thing left for you to patch on top of that is decide some variable names and such if you rather want something else there.

Tectu, 2024-02-01:
There's no harm in doing a `1.5.2` release prior to the GPDS stuff being resolved as you can simply use the newer GPDS version when it's ready, right?

bjornstromberg, 2024-02-01:
not to my knowledge, was more to flesh out any issues so we dont need to do 1.5.3 for some small misstake..

Tectu, 2024-02-01:
Sounds reasonable. Let's complete simulton/gpds#18 first - shouldn't take long anyway :)

Tectu, 2024-02-02:
So regarding a `1.5.2` release...
Commit 0904b8c26877107e5e6b1b35a5bfb031361c6d7a changed the minimum required C++ level from 17 to 20.
That does not really bode well for a patch release. I would be willing to sneak that into a minor release since C++20 is pretty widely adopted and we're not currently relying on features that are less commonly adopted.

But that feels like a pretty silly minor release.

bjornstromberg, 2024-02-02:
one solution is to branch a release-1.5 from tag 1.5.1 and cherrypick n tag on that branch, that way you exclude the bump in c++ level and save it until more fitting minor bump.

bjornstromberg, 2024-02-02:
or i can cherrypick it as a patch for my packaging usage

bjornstromberg, 2024-02-02:
leave the tag, i cherrypicked it as a patch. now bed.

commit 416d9736 (2024-01-31): Replace remaining usage of QSchematic::Item with QSchematic::Items::Item

Reported by @bjornstromberg
Fixes #56