simulton/QSchematic #60: v2: node & wire router

bjornstromberg, 2024-02-03:
add a abstract component that handles placing nodes and drawing the wires, we can have a `dumb` generalized solution that is supplied with QSchematics. 

this is essentialy a way to get rid of positions in saved files.

of course there should be a way to place something by hand or move by hand, but run the router again, and stuff goes back in line, even a way to disable the routings hand have handplaced positions still if that is wanted.

Tectu, 2024-02-03:
Can you elaborate on that a bit more?

bjornstromberg, 2024-02-04:
Right now I'm about to do a reference implementation for my own usage, so there is still alot of unknowns. 

Routings is the info in the netlist and settings for the router is like avoid overlap paths that are not part of same path 

As a visual demo take 6 roundabouts and draw the lanes of the road between them in a netlist. 

The router now moves each node to a sane location with some ranged spacings supplied.

bjornstromberg, 2024-02-04:
do we have any opinions on the definition of a Schematic / Diagram, 
do they have to be static, or are dynamic diagrams allowed? 

in the road / roundabout example, is dynamic flow allowed to be integrated? 
think of a dot (car) going on its way past the available track..

bjornstromberg, 2025-02-20:
> > Right now I'm about to do a reference implementation for my own usage, so there is still a lot of unknowns.
> 
> Did you ever get that up and running?

unfortunately the project that needed this got postponed, not sure when i get allocation of time to revisit/finish this, 
got non-qt, C/C++ work up above the ears for now that got priority,
probably should not complain that the work queue is overflowing, the other way around would be worse... 

the part i have written (code is on a usb stick somewhere) is essentially a separate c++ graph library, 
that says node a is enum{left, right, above, below} of node b where left and right are exclusive, 
and above and below are exclusive, 
it starts by placing the nodes with zero dependencies first and work its way through the graph until its done.

the reason for making this a separate library was that it can support dynamic changes, 
without forcing that logic into QSchematic.
eg. the scene can contain a timeline of changes without actually care about the pixel by pixel changes unless a big change happens 
like `a` moves `above` `b` from `below` etc.

reason for that was to reuse the library as a gui for a custom work-pipeline tool (the road demo stuff noted above), 
which is pretty much a timeline + a schematic, intended to make invisible IT work visible..  

as the graph grows, the actual changes in priority should only be recorded when they move 'enough' not a few pixels here or there.. 

so current timeline is probably `End of Year -25` to not promise to much, 
the big issue is to integrate it nicely with QSchematic, 
as it currently works against Qt GraphicsView level with dummies, looks like crap but worked.

Tectu, 2025-03-16:
I think what we'd truly need/want is some sort of layout engine.
Basically, an interface where we can input a netlist (rather than the full editor config which contains all the positions etc) and then it lays out the nodes in a reasonable way™.

bjornstromberg, 2025-03-17:
Basically it feels like we need to create the test data first and, what do
we expect the input to create for valid outputs.

Without a clear vision on what inputs and outputs upfront are valid and
also invalid, i think we are in a long pingpong match back and forth with
commits to get anywhere...

Working with raw netlist specs is nothing I've done before so that is
probably a minefield for me (or it will be really simple, but not done any
research on that yet..)

bjornstromberg, 2025-03-17:
one solution to getting the netlist formats specified cleanly is to rework the testsuite to actually define the netlist format 

because from the current  [net-tests](https://github.com/simulton/QSchematic/blob/666a1b7bdada5a2db06876e3c76bb3f54d54e932/qschematic/wire_system/test/tests/nets.cpp) i really would expect it to define how the library expects a netlist to be defined.

since the 'scene' as stored in gpds and other formats differ (location/position data etc I've seen you mention before), 
i would say that we should probably start right there... 

what is a netlist?
imho its a list of nodes and edges aka a graph, and since we want Documents (#44) it can also be a hierarchical graph.

so the "layout engine" as i wrote it last year, works its way though the graph in {N, E, S, W} and just pops of any nodes it already has placed from the placement queue that way there was always at-least one node to place if the queue was non-empty. it was very crude, and did not look very good since it only used straight lines. reason for the relative information was that i had an idea to make the graph compress itself and expand on zoom in/out. that was never written though..

this could be refined a lot by just enhancing it to a compass-rose like:
[image]

but it also gives a-lot more possibilities that has to be tested