simulton/QSchematic #65: Build faild when disable QSCHEMATIC_BUILD_STATIC under linux

vowstar, 2024-02-17:
Build faild when disable DQSCHEMATIC_BUILD_STATIC under linux, but enable DQSCHEMATIC_BUILD_STATIC it build success.

OS: Gentoo Linux
GCC: gcc version 13.2.1 20240210 (Gentoo Hardened 13.2.1_p20240210 p13)
QT: QT6

```bash
cmake -B build -G Ninja -DQSCHEMATIC_BUILD_STATIC=OFF -DQSCHEMATIC_BUILD_DEMO=OFF -DQSCHEMATIC_DEPENDENCY_GPDS_TARGET:STRING="gpds::gpds-shared" -DQSCHEMATIC_DEPENDENCY_GPDS_DOWNLOAD=OFF
-- 
-- -------------------------
-- QSchematic configuration:
--   version      : 1.5.1
-- 
--   Build
--     Static     : OFF
--     Shared     : ON
--     Demo       : OFF
-- 
--   Dependencies
--     GPDS
--       Download : OFF
--       Target   : gpds::gpds-shared
-- -------------------------
-- 
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/vowstar/Projects/2023/QSchematic/build
```

```bash
QSchematic git:(master) cmake --build build -j

[1/48] Building CXX object qschematic/...atic-wiresystem-tests.dir/__/net.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/net.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/net.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/net.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/net.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/net.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/net.cpp:1:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/net.hpp:3:10: fatal error: QList: No such file or directory
    3 | #include <QList>
      |          ^~~~~~~
compilation terminated.
[2/48] Building CXX object qschematic/...ic-wiresystem-tests.dir/__/point.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/point.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/point.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/point.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/point.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/point.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/point.cpp:1:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/point.hpp:3:10: fatal error: QPointF: No such file or directory
    3 | #include <QPointF>
      |          ^~~~~~~~~
compilation terminated.
[3/48] Building CXX object qschematic/...tic-wiresystem-tests.dir/__/line.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/line.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/line.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/line.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/line.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/line.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/line.cpp:1:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/line.hpp:3:10: fatal error: QPointF: No such file or directory
    3 | #include <QPointF>
      |          ^~~~~~~~~
compilation terminated.
[4/48] Building CXX object qschematic/...wiresystem-tests.dir/__/__/utils.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/utils.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/utils.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/utils.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/utils.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/utils.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/utils.cpp:1:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/line.hpp:3:10: fatal error: QPointF: No such file or directory
    3 | #include <QPointF>
      |          ^~~~~~~~~
compilation terminated.
[5/48] Building CXX object qschematic/...tic-wiresystem-tests.dir/__/wire.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/wire.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/wire.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/wire.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/wire.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/wire.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/wire.hpp:3,
                 from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/wire.cpp:1:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/point.hpp:3:10: fatal error: QPointF: No such file or directory
[... 28 lines cut ...]
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/settings.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/settings.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/settings.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/__/__/settings.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/settings.cpp
/home/vowstar/Projects/2023/QSchematic/qschematic/settings.cpp:3:10: fatal error: QPointF: No such file or directory
    3 | #include <QPointF>
      |          ^~~~~~~~~
compilation terminated.
[10/48] Building CXX object qschematic...-wiresystem-tests.dir/tests/nets.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/test/tests/nets.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/test/tests/nets.cpp:2:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/test/tests/../../manager.hpp:5:10: fatal error: QObject: No such file or directory
    5 | #include <QObject>
      |          ^~~~~~~~~
compilation terminated.
[11/48] Building CXX object qschematic...-wiresystem-tests.dir/tests/wire.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o -c /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/test/tests/wire.cpp
In file included from /home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/test/tests/wire.cpp:2:
/home/vowstar/Projects/2023/QSchematic/qschematic/wire_system/test/tests/../../manager.hpp:5:10: fatal error: QObject: No such file or directory
    5 | #include <QObject>
      |          ^~~~~~~~~
compilation terminated.
[13/48] Building CXX object qschematic...c-wiresystem-tests.dir/test_main.cpp.o
ninja: build stopped: subcommand failed.
```

Tectu, 2024-02-17:
I'm currently unable to reproduce this. Can you try performing a clean build (destroying the build directory and starting over)?
If the problem persists, can you please share the complete terminal log from start to finish from your clean build?

vowstar, 2024-02-18:
Thank you for your patient response. In order to conveniently reproduce this issue and to demonstrate its prevalence across most Linux distributions, I have constructed a minimal reproduction environment using Docker, based on the ``debian:bookworm`` distribution. To replicate the issue, simply ensure Docker is installed, save the following content as a ``Dockerfile``, and then run ``docker build -t issue-65``. This will reproduce the problem.

Dockerfile:

```dockerfile
FROM debian:bookworm
LABEL maintainer="Huang Rui <vowstar at gmail.com>"

## Configure packages
RUN DEBIAN_FRONTEND=noninteractive apt-get update -qq && \
    DEBIAN_FRONTEND=noninteractive apt-get install -yq \
        '^qt6-.*-dev' \
        build-essential \
        cmake \
        ninja-build \
        git \
    && \
    apt-get clean && rm -rf /var/cache/apt/archives && rm -rf /var/cache/apt/lists

## Clone GPDS
RUN git clone https://github.com/simulton/gpds.git

## Build and Install GPDS
RUN cd gpds && \
    cmake -B build -G Ninja \
        -DCMAKE_BUILD_TYPE=Release \
        -DCMAKE_INSTALL_PREFIX=/usr/local \
		-DGPDS_BUILD_STATIC=OFF \
		-DGPDS_BUILD_SHARED=ON \
		-DGPDS_BUILD_TESTS=OFF \
		-DGPDS_BUILD_EXAMPLES=OFF \
    && \
    cmake --build build --target install

## Clone QSchematic
RUN git clone https://github.com/simulton/QSchematic.git

## Build and Install QSchematic
RUN cd QSchematic && \
    cmake -B build -G Ninja \
    -DQSCHEMATIC_BUILD_STATIC=OFF \
    -DQSCHEMATIC_BUILD_DEMO=OFF \
    -DQSCHEMATIC_DEPENDENCY_GPDS_TARGET:STRING="gpds::gpds-shared" \
    -DQSCHEMATIC_DEPENDENCY_GPDS_DOWNLOAD=OFF \
    && \
    cmake --build build --target install
```

Command to reproduce:

```bash
docker build -t issue-65
```

The result:

```bash
Sending build context to Docker daemon  3.072kB
Step 1/8 : FROM debian:bookworm
 ---> 52f537fe0336
Step 2/8 : LABEL maintainer="Huang Rui <vowstar@gmail.com>"
 ---> Using cache
 ---> 0f225ba83273
Step 3/8 : RUN sed -i 's|http://deb.debian.org|http://mirrors.tuna.tsinghua.edu.cn|g' /etc/apt/sources.list.d/debian.sources &&     sed -i 's|http://security.debian.org|http://mirrors.tuna.tsinghua.edu.cn|g' /etc/apt/sources.list.d/debian.sources &&     sed -i 's|main$|main contrib non-free|g' /etc/apt/sources.list.d/debian.sources
 ---> Using cache
 ---> 35b588656e02
Step 4/8 : RUN DEBIAN_FRONTEND=noninteractive apt-get update -qq &&     DEBIAN_FRONTEND=noninteractive apt-get install -yq         '^qt6-.*-dev'         build-essential         cmake         ninja-build         git     &&     apt-get clean && rm -rf /var/cache/apt/archives && rm -rf /var/cache/apt/lists
 ---> Using cache
 ---> eb4c600067ef
Step 5/8 : RUN git clone https://github.com/simulton/gpds.git
[... 114 lines cut ...]
    5 | #include <QObject>
      |          ^~~~~~~~~
compilation terminated.
[10/49] Building CXX object qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/nets.cpp.o -c /QSchematic/qschematic/wire_system/test/tests/nets.cpp
In file included from /QSchematic/qschematic/wire_system/test/tests/nets.cpp:2:
/QSchematic/qschematic/wire_system/test/tests/../../manager.hpp:5:10: fatal error: QObject: No such file or directory
    5 | #include <QObject>
      |          ^~~~~~~~~
compilation terminated.
[11/49] Building CXX object qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o
FAILED: qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o 
/usr/bin/c++    -MD -MT qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o -MF qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o.d -o qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/tests/wire.cpp.o -c /QSchematic/qschematic/wire_system/test/tests/wire.cpp
In file included from /QSchematic/qschematic/wire_system/test/tests/wire.cpp:2:
/QSchematic/qschematic/wire_system/test/tests/../../manager.hpp:5:10: fatal error: QObject: No such file or directory
    5 | #include <QObject>
      |          ^~~~~~~~~
compilation terminated.
[12/49] Automatic MOC for target qschematic-shared
[13/49] Building CXX object qschematic/wire_system/test/CMakeFiles/qschematic-wiresystem-tests.dir/test_main.cpp.o
ninja: build stopped: subcommand failed.
The command '/bin/sh -c cd QSchematic &&     cmake -B build -G Ninja     -DQSCHEMATIC_BUILD_STATIC=OFF     -DQSCHEMATIC_BUILD_DEMO=OFF     -DQSCHEMATIC_DEPENDENCY_GPDS_TARGET:STRING="gpds::gpds-shared"     -DQSCHEMATIC_DEPENDENCY_GPDS_DOWNLOAD=OFF     &&     cmake --build build --target install' returned a non-zero code: 1

```

Tectu, 2024-02-19:
Thank you for providing this information. I am able to reproduce it.
Should be able to provide a fix very soon.

Tectu, 2024-02-19:
Can you check whether the latest `master` branch is working for you?
We can provide a new release afterwards to help your packaging efforts.

closed 2024-02-19 by Tectu

vowstar, 2024-02-19:
Thank you so much for your prompt response and for addressing the issue. I have tested the latest master branch, and I'm happy to confirm that the problem has been **successfully resolved**.

Your swift action is greatly appreciated and will significantly aid in our packaging efforts. Looking forward to the new release!

Tectu, 2024-02-19:
Thank you for confirming that it's working now.

Release `1.6.0` has just landed - Happy packaging!

commit 66a7cb1b (2024-02-19): cmake: Fix qschematic-wiresystem-test target

This was failing when QSCHEMATIC_BUILD_STATIC was set to OFF.

Reported by @vowstar.
Fixes #65