simulton/QSchematic #69: How to transport data based on signal/slot bewtten two connector?

maobaoqi, 2024-04-28:
Hi, I clone this project and compiled,  I think this is a great work for learning qt graphics flow.

I try to use this project to create a algorithm simulator system. There is a basic question, how to transport data between two demo operator using connector, for example, one connector is INPUT and another is OUTPUT, demo operator contains a core
process function.

Maybe the key point is that how to auto connect one connector signal to another connector slot when draw the fancywire. 
Any good and efficient ideas? thanks !

Tectu, 2024-04-28:
This library is a graphics/view framework only. Any sort of actual logic needs to be implemented by the consuming application.

The `Netlist` (and `NetlistGenerator`) serve as the interface to that logic.
If you need more fine grained control you can implement your own `NetlistGenerator` or directly tap the `wire_system` API.

maobaoqi, 2024-04-29:
Thanks for the reply. Under docs folder, there are some screenshot pictures, screenshot_04.png screenshot_05.png screenshot_06.png contains icon and are beautiful, how to add icon for one node?

Tectu, 2024-04-29:
Overload the `QSchematic::Items::Node::paint()` function. The example/demo shows how to do this.

Please open a new issue with matching title if you have more questions :)

closed 2024-04-29 by Tectu