simulton/QSchematic #72: Cmake error

AmerZater, 2024-12-25:
I am usign Cmake (3.27.7) 
Cmake Generator Ninja

my cmakeLists:
                      include(FetchContent)
                      
                      # Set QSchematic CMake options
                      set(QSCHEMATIC_BUILD_DEMO               OFF          CACHE INTERNAL "")
                      set(QSCHEMATIC_DEPENDENCY_GPDS_DOWNLOAD ON          CACHE INTERNAL "")
                      set(QSCHEMATIC_DEPENDENCY_GPDS_TARGET  "gpds-shared" CACHE INTERNAL "")
                      
                      # Fetch QSchematic
                      FetchContent_Declare(
                          qschematic
                          GIT_REPOSITORY https://github.com/simulton/qschematic
                          GIT_TAG        main
                      )
                      FetchContent_MakeAvailable(qschematic)
                      
                      
                      option(LINK_INSIGHT "Link Qt Insight Tracker library" ON)
                      option(BUILD_QDS_COMPONENTS "Build design studio components" ON)
                      
                      project(BF_ConfigServerApp LANGUAGES CXX)
                      
                      set(CMAKE_AUTOMOC ON)
                      
                      find_package(Qt6 6.2 REQUIRED COMPONENTS Core Gui Qml Quick)
                      
                      
                      if (Qt6_VERSION VERSION_GREATER_EQUAL 6.3)
                          qt_standard_project_setup()
                      endif()
                      
                      qt_add_executable(BF_ConfigServerApp src/main.cpp)
                      
                      qt_add_resources(BF_ConfigServerApp "configuration"
                          PREFIX "/"
                          FILES
                              qtquickcontrols2.conf
                      )
                      
                      target_link_libraries(BF_ConfigServerApp PRIVATE
                          Qt6::Core
                          Qt6::Gui
                          Qt6::Qml
                          Qt6::Quick
                          qschematic::qschematic-static
                      )

i had this cmake error:

**[cmake] CMake Error:
[cmake]   Running
[cmake] 
[cmake]    '/home/ubuntu/Qt/Tools/Ninja/ninja' '-C' '/home/ubuntu/Downloads/build-QSchematic-main-WebAssembly_Qt_6_5_3_single_threaded-Debug' '-t' 'recompact'
[cmake] 
[cmake]   failed with:
[cmake] 
[cmake]    ninja: error: [build.ninja:1106](javascript:void(0);): multiple rules generate _deps/gpds-build/lib/libgpdsd.a [-w dupbuild=err]**

Tectu, 2024-12-25:
Can you try building the GPDS dependency separately just to verify that it itself builds & installs correctly?

AmerZater, 2024-12-26:
Hello Mr. Bodenmann
I tried to build  GPDS  alone.. it gave me the next error in concepts.hpp :

/home/ubuntu/Downloads/gpds-main/lib/include/gpds/concepts.hpp:1:10: fatal error: concepts: No such file or directory
    1 | #include <concepts>
      |          ^~~~~~~~~~

I am using Qt 6.5 LTS LIBs

Tectu, 2024-12-26:
Ignore the QSchematic issues until you are able to successfully build/install GPDS.

Are you sure that you are using a C++20 capable compiler? It looks like it doesn't manage to `#include <concepts>` which is a C++20 feature.

Tectu, 2025-02-19:
Closing this due to submitter inactivity.

closed 2025-02-19 by Tectu