simulton/QSchematic #73: running tests via ctest fails with 1.6.0 but not 1.5.1 (no tests actually runs in either of them..)

bjornstromberg, 2025-03-16:
so yea I've been kinda busy with a massive backlog of other stuff now I got around to bump my alpine package script to 1.6.0 (was using 1.5.1 + a patch earlier)

anyway what i found was kinda interesting.. we do have a testsuite for the wire-system but because `I never actually verified that it was run`, I did not catch that earlier..

```
> packaging 1.5.1 for alpine linux 3.21 (Qt-6.7) 
-- snip --
[100%] Linking CXX executable qschematic-wiresystem-tests
[100%] Built target qschematic-wiresystem-tests
*********************************
No test configuration file found!
*********************************
Usage

  ctest [options]

>>> qschematic: Entering fakeroot...
-- Install configuration: "RelWithDebInfo"
-- snip --
```

```
> packaging 1.6.0 for alpine linux 3.21 (Qt-6.7) 
-- snip --
[100%] Linking CXX executable qschematic-wiresystem-tests
[100%] Built target qschematic-wiresystem-tests
Internal ctest changing into directory: /**path**/alpine-upgrades/simulton-qschematic/src/build
Test project /**path**/alpine-upgrades/simulton-qschematic/src/build
No tests were found!!!
Errors while running CTest
>>> ERROR: simulton-qschematic: check failed
>>> simulton-qschematic: Uninstalling dependencies...
-- snip --
```

so what diffs more than the versions?

```
> 1.5.1 build scripts check() function
check() {
        cd $srcdir/build
        CTEST_OUTPUT_ON_FAILURE=TRUE ctest
}
```

```
> 1.6.0 build scripts check() function 
check() {
       ctest --test-dir $srcdir/build --output-on-failure
}
```

if i run 1.6.0 with the 1.5.1 check function, the ctest output does not kill the build. so its a silent error by ctest that was overlooked.

so we're likely missing `add_test(NAME test-exec COMMAND test-exec )` in `https://github.com/simulton/QSchematic/blob/1.6.0/qschematic/wire_system/test/CMakeLists.txt`

I'll see if i can fix this quickly, what good is a test that does not run when you expect it to have done so...

bjornstromberg, 2025-03-16:
funny enough it looks like APKBUILD (alpine packaging system) is harder than the ci on the no-tests run part..

[ci actions log](https://github.com/bjornstromberg/QSchematic/actions/runs/13883003446/job/38843981593)

bjornstromberg, 2025-03-16:
@Tectu this one is ready to be merged. now CI runs via ctest, pretty much a trivial cmake change.

Tectu, 2025-03-16:
Thanks - I'll have a look.

Depending on what you're up to you might want to have a look at the current main branch. It especially address the issue with the previously statically rendered background (huge memory requirement & pixelated when zooming in). It's now rendered dynamically.

I keep debating whether I should make a `1.7.0` release or just go for `2.0.0`. The latter would be more correct for sure.

bjornstromberg, 2025-03-16:
nice that the ugly static pix-map is gone on master, that sure sounds worthy of a 1.7.0 tag

I would wait with 2.0.0 until we get the compilation firewall idiom and some of the bigger redesigns like the Document, 
mostly since it would look strange to have a big redesign after 2.0 rather than before 2.0 not that it really maters.

I have queued up a few issue branches on my github fork where I at-least intend to get some work done this year to get closer to the v2.0 release.

one of the big things by using ctest now is that we can separate the test suites where issue-x can get its own executable if its a larger piece of work, which needs to be integrated against other stuff (Document for example).

Tectu, 2025-03-16:
Well, my rationale is that the new background thing breaks API as I couldn't be bothered supporting the old, clearly wrong/inferior background handling and marking it as deprecated.

Typically, a minor release wouldn't introduce breaking API changes of this magnitude IMHO.
We can always just do 2.0.0 now and then have a 3.0.0 release with the other stuff that is still in the pipeline.

That being said, I didn't really worry about API breakage so far either.

Thoughts?

Also, I am supportive of the CTest stuff. I am using CTest extensively nowdays.

Tectu, 2025-03-16:
Ah, it's almost definitely going to be a 2.0.0 release. I already made other changes such as dropping Qt5 support and bumping to C++23.

bjornstromberg, 2025-03-16:
> Well, my rationale is that the new background thing breaks API as I couldn't be bothered supporting the old, clearly wrong/inferior background handling and marking it as deprecated.

sounds correct with 2.0 then
 
> Typically, a minor release wouldn't introduce breaking API changes of this magnitude IMHO. We can always just do 2.0.0 now and then have a 3.0.0 release with the other stuff that is still in the pipeline.

I'm good with aiming for a 3.0 release rather than a 2.2 or such.

> That being said, I didn't really worry about API breakage so far either.

the idea with v2 was to get things in shape good enough to actually dare to use it as a shared library, 
but I can live with that being v3 too..

> Thoughts?

go for v2 now and rename the 1-milestone to v3 its just cosmetics anyway

> Also, I am supportive of the CTest stuff. I am using CTest extensively nowdays.

it sure makes life easer...

> Ah, it's almost definitely going to be a 2.0.0 release. I already made other changes such as dropping Qt5 support and bumping to C++23.

that sure also qualifies for 2.0

closed 2025-03-16 by Tectu

Tectu, 2025-03-16:
`v2.0.0` release done.

Thanks!

PR #74 (2025-03-16): enable ctest: Fixes issue #73
fixes #73 by allowing for ctest to run from the build-root dir
commit a72b659d (2025-03-16): CTest: Use CTest for running test suites

Fixes #73